micronucleus/micronucleus #182: interest in an asm version of micronucleus

nerdralph, 2020-08-20:
I've written a few tiny bootloaders in AVR assembler, and am thinking of writing a USB bootloader.  If there's interest in an optimized version of micronucleus in pure assembler for the avr25 architecture, I'd be willing to write it.  I think I can get the t85/16.5Mhz version down to 1kB.
If there's not interest in an asm branch for micronucleus, I may just copy it to use as a base, since it already has a working programming tool.

nerdralph, 2020-08-23:
Thanks for the feedback Tim,

Although it doesn't seem to be very popular, I prefer to do most of my embedded programming in asm, particularly on small 8-bit platforms like 8051 and AVR.  I find it much easier to write the asm code I want, rather than trying to cajole avr-gcc into generating the code that I want.

I agree with your plan; that's pretty much what I've started doing.  t85_default will be my main testing target, with t85_aggressive being secondary.

I shouldn't have any questions on the technical side, as low-speed USB 1.1 is a rather simple protocol when limited to EP0 control packets.  Where I may have some questions is to tap your knowledge of how people are actually using micronucleus.  Sometimes functionality that has no value to me is important to the average user.

Regarding CI,  it's a good suggestion.  I'll probably use github actions for CI on my fork.  Just something simple to make sure everything builds clean.

nerdralph, 2020-08-24:
I just got t85_aggressive down to 1342 bytes, below the next page boundary.
https://github.com/nerdralph/micronucleus-firmware/tree/asm

I also improved the sync timing window from -1/+11c to -3/+7c per JK transition.

ArminJo, 2020-08-24:
Congrats 🥇

cpldcpu, 2020-08-26:
Nice! Every byte is hard work at this point 👍

nerdralph, 2020-08-27:
Making tweaks to the v-USB asm code is a bit trickier than it should be.  Some places the cycle numbers in the comments are at the start of the instruction, and in others it is at the end.  I think Christian started off using end of instruction counts (usbdrvasm12.inc) and switched to the start of instruction  in the other versions.  Going with the start of instruction makes more sense to me since skip and branch instructions take 1 or 2 cycles.
I've also noticed some mistakes in the cycle counts.  The push on line 96 actually ends with cycle 12.
https://github.com/micronucleus/micronucleus/blob/master/firmware/usbdrv/usbdrvasm12.inc#L96
The cycle numbers for the next two push instructions are off by 2, and then from line 100 they are correct.

nerdralph, 2020-08-27:
I've added a m88p config, and finished converting asm12.inc to conform to the avr-libc calling convention.  This makes it easy to run micronucleus on a USBasp.  I've switched to testing with the 12MHz v-USB with external xtal.  No variations of the 16.5MHz version is completely stable on my PC, making it difficult to test changes I make to micronucleus.  I suspect most USB hub controllers don't implement a PLL with a timing range of +/- 1.5% for low-speed devices as required by the spec.  Tim, it looks like you figured this out several years ago:
https://forums.obdev.at/viewtopic.php?t=8819

In addition to the code size optimizations I'm doing, I think there is still room to improve reliability.  I think OSCCAL tuning doesn't always pick the best value (sometimes being off by 1 step), and I think I can reduce the code size.  In picoboot I've implemented automatic OSCCAL tuning in 20 instructions.
https://github.com/nerdralph/picoboot-lib/blob/master/pb-lib.S#L107

nerdralph, 2020-08-27:
I just pushed an update to usbdrv.c that leaves out unused USB requests (ones mn should never see, or ignores).  t85_default is now 1438 bytes and t85_aggressive is 1292.

nerdralph, 2020-08-27:
I'm catching some big fish today.  Since the configuration descriptor is now only 9 bytes, wLength no longer needs to be checked for any of the messages mn receives.  t85_aggressive is now under the next page boundary (1280) at 1274 bytes.

ArminJo, 2020-08-27:
👍 
If you push the hex files to your repo, I can make some independent and intermediate tests for your t85_default and t85_aggressive versions.

nerdralph, 2020-08-27:
> 
> 
> +1
> If you push the hex files to your repo, I can make some independent and intermediate tests for your t85_default and t85_aggressive versions.

As you wish.
https://github.com/nerdralph/micronucleus-firmware/tree/asm/firmware/releases

nerdralph, 2020-08-28:
I removed more dead code from usbdrv.c, and got past the next page boundary for t85_default: 1398 bytes.

cpldcpu, 2020-08-28:
>  I suspect most USB hub controllers don't implement a PLL with a timing range of +/- 1.5% for low-speed devices as required by the spec. Tim, it looks like you figured this out several years ago:
> https://forums.obdev.at/viewtopic.php?t=8819

Oh indeed, I did not even remember that. I guess I was trying to corroberate the claims in the V-USB documentation regarding the stability of the individual implementations for different clockspeeds. Not all of the claims seem to make sense and none are backed up with data.

I am a bit surprised, however, that you see issues with 16.5 MHz. I always regarded this as a stable variant.

> In addition to the code size optimizations I'm doing, I think there is still room to improve reliability. I think OSCCAL tuning doesn't always pick the best value (sometimes being off by 1 step), [...]

Keep in mind that some ATtinys have a non-monotonic encoding of OSCCAL. Therefore the linear search ("neighbourhood search") should not be skipped. Also, an timeout conditition is handy to prevent deadlocks.

nerdralph, 2020-08-29:
> Oh indeed, I did not even remember that. I guess I was trying to corroberate the claims in the V-USB documentation regarding the stability of the individual implementations for different [...]
The reference to +/- 1.5% is correct. 7.1.11 - data signalling pg 159
"For low-speed functions, the required data-rate whentransmitting (TLDRATE) is 1.50 Mb/s ±1.5% (15,000 ppm)."
http://sdphca.ucsd.edu/lab_equip_manuals/usb_20.pdf

However I'm convinced that testing with a low-speed device outside of +/- 0.25% timing is not part of the certification process, and the DPLL in many modern hubs doesn't have the range to vary by 1.5%.  Hence the most stable v-usb devices are those that use an external xtal, such as the USBasp and USBtinyISP.

nerdralph, 2020-08-29:
Today I started converting the usbdrv.c ProcessRx code to asm.  t85_aggressive is now below 19 pages, at 1212 bytes.
https://github.com/nerdralph/micronucleus-firmware/blob/asm/firmware/releases/t85_aggressive20200828.hex

next I'll work on BuildTx.

nerdralph, 2020-08-30:
I found some easy size optimizations in the C code by using fixed registers for usbMsgPtr and usbMsgLen.  Those two changes save a total of 36 bytes, so now t85_default is down to 1340 bytes.
https://github.com/nerdralph/micronucleus-firmware/blob/asm/firmware/releases/t85_default20200830.hex

nerdralph, 2020-08-30:
I finished converting usbBuildTxBlock to asm, and now t85_aggressive is 1146 bytes.
https://github.com/nerdralph/micronucleus-firmware/blob/asm/firmware/releases/t85_aggressive20200830.hex

cpldcpu, 2020-08-31:
👍  Yeah, it's just a lot of nested functions that can be flattened.

nerdralph, 2020-09-01:
Today I decided to do some investigation into the reliability problems I'm having with micronucleus on a t85 at 16/16.5MHz.  While it's well known OSCCAL tuning on the t85 doesn't get much closer than +-0.5%, what I found is the frequency is only stable to within +/- 0.5% over short intervals (~1s).  Here's the program and a sample output.
https://github.com/nerdralph/debugSerial/blob/master/examples/clockProfiler/clockProfiler.ino
```
bootup OSCCAL: 0x58
external clock signal detected
testing OSCCAL 0x54...
avg cycles/ms:15732, high-low:15776-15648
testing OSCCAL 0x55...
avg cycles/ms:15811, high-low:15904-15792
testing OSCCAL 0x56...
avg cycles/ms:15930, high-low:16016-15904
testing OSCCAL 0x57...
avg cycles/ms:16062, high-low:16176-16032
testing OSCCAL 0x58...
avg cycles/ms:16061, high-low:16160-16032
testing OSCCAL 0x59...
avg cycles/ms:16215, high-low:16288-16176
testing OSCCAL 0x5A...
avg cycles/ms:16348, high-low:16432-16288
testing OSCCAL 0x5B...
avg cycles/ms:16484, high-low:16576-16448
testing OSCCAL 0x5C...
avg cycles/ms:16537, high-low:16640-16512
```

cpldcpu, 2020-09-01:
Htat seems to be a surprisingly strong effect. Since the trend is quite monotonic: Could this be due to self-heating?
It could be quite interesting to monitor temperature and VDD using the internal ADC at the same time. 
Maybe some information could be extracted to trigger re-sync?

nerdralph, 2020-09-01:
The resolution of the high/low count was only 16 cycles as I was only sampling 1 phase with a 8-cycle sampling loop and then multiplying by 2.  I updated it to sample both clock phases of the external clock.  I'm using the 1kHz output of my Rigol DS1054Z which is extremely stable and accurate to a few ppm.  I also reduced the sampling window from 1s to 250ms, as I think it starts getting unreasonable if you were to retune OSCCAL more than 4x/s.  
I tried it on a t13 (downclocked to 8M) as that's what I happen to have setup this am.  I'll try an actual t85 later, but I don't expect it to be much different since I've found the RC oscillators on both chips behave similarly.
```
testing OSCCAL 0x49
avg cycles/ms:7605, high-low: 7640-7592
testing OSCCAL 0x4A
avg cycles/ms:7690, high-low: 7728-7680
testing OSCCAL 0x4B
avg cycles/ms:7790, high-low: 7816-7768
testing OSCCAL 0x4C
avg cycles/ms:7883, high-low: 7912-7856
testing OSCCAL 0x4D
avg cycles/ms:7974, high-low: 8000-7952
testing OSCCAL 0x4E
avg cycles/ms:8068, high-low: 8096-8048
testing OSCCAL 0x4F
avg cycles/ms:8157, high-low: 8200-8144
testing OSCCAL 0x50
avg cycles/ms:8165, high-low: 8192-8128
testing OSCCAL 0x51
avg cycles/ms:8282, ⸮igh-low: 8304-8240
```

Since the resolution of the OSCCAL steps and the clock instability are cumulative, that means you can't count on much better than +/- 1% timing accuracy on a t85.

nerdralph, 2020-09-01:
I just noticed something that *might* explain some of the unreliability I'm seeing while running on the internal RC oscillator:
https://github.com/micronucleus/micronucleus/blob/master/firmware/osccalASM.S#L153

I haven't been testing with a dedicated bus; there's a low-speed mouse with a 8ms poll interval on the same bus. I think it's too limiting to require a dedicated USB bus for micronucleus, as many low-end PCs and laptops only have two.  As well, you'd need to explain to most users how to determine which physical USB ports share one bus.

I think before I work on converting the main loop to asm, I'm going to write a new calibration function.  What I use in picoboot is a proportional algorithm; a larger time difference results in a larger change to OSCCAL.  I think I can get OSCCAL to the optimal value with 3-4 empty frames, and reliably discard timing from non-empty frames.

cpldcpu, 2020-09-01:
These collision are not as much of an issue as one might think. Only old (1.1?) USB hubs forward all traffic to every port. Anything more modern should not present this issue. AFAIR the fix was introduced by shay based on his observations on a 1.1 hub.

cpldcpu, 2020-09-01:
Regarding the clock: There certainly could be some low frequency noise that leads to local variation. Have you checked whether there is any consistent drift? That would be the bigger issue.

nerdralph, 2020-09-01:
> 
> These collision are not as much of an issue as one might think. Only old (1.1?) USB hubs forward all traffic to every port. Anything more modern should not present this issue. AFAIR the fix was introduced by shay based on his observations on a 1.1 hub.

It's certainly not limited to USB 1.1 hubs.  The Intel USB 2.0 controller in my PC forwards the low-speed traffic to every port.  I can see the USB IN token packets every 8th frame on a different port when I pull up DMINUS.

nerdralph, 2020-09-01:
> 
> 
> Regarding the clock: There certainly could be some low frequency noise that leads to local variation. Have you checked whether there is any consistent drift? That would be the bigger issue.
Since micronucleus runs immediatly after reset, and recalibrates OSCCAL each time, so long-term drift is not a concern.  Short-term variations in clock stability are what I was interested in quantifying.

FWIW, I save the tuned OSCCAL value in picoboot the first time it is run, so the tuning code doesn't take any space in the bootloader.  The timing margins for a UART are much wider (+/- 5%), so even if it changed by percent or so weeks later due to temperature, serial communications will still be error-free.  Looking at figure 22-41 of the t85 datasheet, going from 20C to 25C should lead to a frequency increase of only .16%, so the same technique may even be feasible for micronucleus when used in a "normal" indoor setting.

cpldcpu, 2020-09-02:
Interesting, I never observed that was any of my rigs.  The problem with the solution currently in MN is that it will extend the calibration time if traffic is present. A better solution should just ignore the data without losing track of the frame timing. One could use a hold-off period, very elaborate cycle counting or use the hardware timer right away.

nerdralph, 2020-09-02:
> 
> Interesting, I never observed that on any of my rigs. The problem with the solution currently in MN is that it will extend the calibration time if traffic is present. A better solution should just ignore the data without losing track of the frame timing. One could use a hold-off period, very elaborate cycle counting or use the hardware timer right away.

That's a good idea.   Is something like this what you had in mind?
https://github.com/nerdralph/micronucleus-firmware/blob/asm/firmware/osccal.S

cpldcpu, 2020-09-02:
Ah, a fixed point iteration. That's a very good idea!  It should also be much faster than the old osccal and therefore not create issues with win120.

Did you check if this also works with the split-OSCAL-range ATinies?

Btw, isnt't the rjmp in line 27 obselete? 

You could save a lot of instructions by adding 16 in line 59 and instead shift down by 8 bit -> just use high byte.

nerdralph, 2020-09-02:
Great suggestions Tim.
AFAIK, Atmel's factory calibration on the tX5 always uses the low range below 128.  As I mentioned earlier, I use a similar technique in picoboot's autotuner, and it works well on the t85.

I've implemented almost all of your suggestions, as well as adjusted the OSCCAL step factor to 0.667% as that's closer to the average change on the t85.  It's now down to 42 bytes, so maybe that's a sign to stop there. :-)  Though I think you are right about initializing countH:countL with a negative.

I couldn't figure out how I can get rid of one of the RETs until I started thinking about rearranging the order of the functions.  Are you suggesting I put countFrame between tuneOsccal and tuneOnce?

cpldcpu, 2020-09-03:
This is fun, looks really nice now.

You could remove the RET line line 37. It will increase the runtime by one SOF, but otherwise it will do no harm. Too much compromise maybe?

nerdralph, 2020-09-03:
> 
> 
> This is fun, looks really nice now.
> 
> You could remove the RET line line 37. It will increase the runtime by one SOF, but otherwise it will do no harm. Too much compromise maybe?

Since you enjoy optimizing asm, I'd love to see if you could squeeze more than a couple bytes out of picoboot:
https://github.com/nerdralph/picoboot-lib/blob/master/pb-lib.S

Regarding the line 37 ret, since tuneOnce is called 3 times, having it fall through to countFrame would mean 3 extra SOFs, not just 1.  I even thought of sacrificing some accuracy on the count (less than 1 bit) by dropping the clr countL.  However that goes against the primary goal of improved accuracy/reliability.

cpldcpu, 2020-09-03:
Move line 30 to before line 25? then countframe is only called twice may need to adjust for the timing.

I will have a look at picoboot

nerdralph, 2020-09-03:
> 
> Move line 30 to before line 25? then countframe is only called twice may need to adjust for the timing.
> 
I had previously thought about timing the frames back-to-back, but went for the simple option at first.  What I like about your suggestion is that it could actually cut the number of frame times required to 4 or 5 from 6.

>
> I will have a look at picoboot
>
Thanks.

nerdralph, 2020-09-03:
> I had previously thought about timing the frames back-to-back, but went for the simple option at first. What I like about your suggestion is that it could actually cut the number of frame times [...]

I moved the two countFrame calls to the beginning of tuneOsccal, and let tuneOnce fall through to countFrame.  That not only saves the ret instruction, it reduces the number of SOFs needed from 6 to 5.  And now I correct for the countFrame overhead by initializing countL with the accumulator fraction instead of 0.  I think we've optimized it as much as possible now.

nerdralph, 2020-09-04:
I just tested a t84a, and the performance of it's RC oscillator is night and day compared to the t85.   The 8-cycle resolution of my timing is limiting the ability to measure the stability of the clock.
bootup OSCCAL: 0x9A
external clock detected
testing OSCCAL 0x96, avg cycles/ms:7914, high-low: 7920-7912
testing OSCCAL 0x97, avg cycles/ms:7954, high-low: 7960-7952
testing OSCCAL 0x98, avg cycles/ms:7993, high-low: 8000-7992
testing OSCCAL 0x99, avg cycles/ms:8036, high-low: 8040-8024
testing OSCCAL 0x9A, avg cycles/ms:8071, high-low: 8072-8064
testing OSCCAL 0x9B, avg cycles/ms:8107, high-low: 8112-8104
testing OSCCAL 0x9C, avg cycles/ms:8146, high-low: 8152-8136
testing OSCCAL 0x9D, avg cycles/ms:8184, high-low: 8192-8184
testing OSCCAL 0x9E, avg cycles/ms:8224, high-low: 8232-8216

cpldcpu, 2020-09-04:
Interesting. I recall reading somewhere that the t84A had an improved oscillator compared to the non-A version.

nerdralph, 2020-09-04:
Here's what the OSCCAL curve looks like.
[image]

cpldcpu, 2020-09-04:
The A version seems to be a shrink with updated oscillator. The old from the non-a version looks quite different:

[image]

nerdralph, 2020-09-04:
Today I did some testing with a t88 configured to tune OSCCAL to 12Mhz, and it was failing to enumerate.  I added some debug code to output the clock/1000, and found it was getting tuned to 12.5Mhz.   I think the tuning bug has been there for years, as I had tried to get the t88 working several years ago but had moved on to other projects after I couldn't get it working.
Unfortunately the new tuneOsccal code isn't working either, but I've only done a couple simple tests.  I should have it debugged and working over the next 24 hours.

nerdralph, 2020-09-05:
There were a couple basic bugs in tuneOsccal that I found yesterday.  Thinks like both scratch and shift being defined as r23 (doh!).

The most serious bug took me a while to find.  I created a basic realtime debug log with a 6mbps UART so I could log the counter values from OSCCAL.  With the debug logging in place, I realized tuneOsccal was getting called dozens of times because mn tries to tune during bus reset, not after.  That bug was fixed with an extra two instructions to wait for the end of reset.
It's close to working correctly, with the last issue being a 0.5% difference in frame length from what I expected.

nerdralph, 2020-09-05:
With 2 hours to spare in my 24hr target, I have success!  I have a t88 running rock solid.  I'm using a USBasp PCB since they are a cheap tqfp AVR USB development board.  I use a red LED soldered to pads intended for an 0805 polyfuse as a combined power LED and 3.35V "regulator" since it has a vF of 1.65V at 5-10mA.  Running at 3.35V means I don't need any zeners on UDM/UDP, and I use a 7.5k pullup to 5V on UDM.  I've created a new config for it called t88_asp.
[image]

I've tried multiple connect/disconnects, --erase-only, flashing upgrade.hex, and have had zero errors so far.  With LED blink it's 1322 bytes.  I'll do a bit more testing with it tonight, and tomorrow I plan to test with the t84a.

nerdralph, 2020-09-06:
Upgrade wasn't actually working, it was just reporting success but not writing anything because I had forgot to enable the SELFPRGEN fuse.  After enabling the fuse, I'm getting unreliable results flashing new code using the bootloader.  Sometimes I get flash write error -1, other times I get no errors reported, but the code doesn't seem to have updated.  I'm using OSCCAL_RESTORE_DEFAULT and OSCCAL_SLOW_PROGRAMMING.

cpldcpu, 2020-09-06:
Nice! Do you mean Mega88 instead of tiny88? Because there are some tiny88 development boards out there (TINY-MH88) that seem to work fine. I guess all the ATMega-support was developed on boards with crystals.

nerdralph, 2020-09-06:
No, its a t88.  They have the same monotonically increasing OSCCAL curve as
the t84a.  They don't support an xtal, so the MH88 boards must use a smd
can oscillator.  According to the datasheet 12MHz is the top speed, so
running at 16 would be a slight overclock.  Unlike most tinies, it has
hardware SPI and i2c.
https://www.microchip.com/wwwproducts/en/ATTINY88

I think dropping the speed to 8MHz and then pumping it back up to 12 each
time the main loop runs could be creating harmonic oscillations in the
clock, so I'm going to change it to only change speed when writing or
erasing flash.  I also may pick a sacrificial t88 to test flash operations
with the RC oscillator at 12MHz to see if it works.  I've seen posts from
some paranoid old farts on AVRfreaks warning that you can brick your AVR
doing this, but I've never seen anyone test it.

nerdralph, 2020-09-06:
After modifying the SLOW_PROGRAMMING code to only slow the clock during erase/program, the stability seems better with the t88.  Not quite perfect, as I got a recoverable error once during testing:
```
>> Eep! Connection to device lost during erase! Not to worry
>> This happens on some computers - reconnecting...
>> Reconnected! Continuing upload sequence...
```

Upgrade isn't working for me though.  Even though the code sucessfully uploads, the old bootloader doesn't get replaced.  I think this is a bug introduced in the micronucleus-firmware fork, as the upgrade for 2.04 worked fine when I tried it on a t85.   Since flashing and erasing with the mn tool works fine,  I'll leave the upgrade debugging to someone else.

nerdralph, 2020-09-06:
I decided to have some fun today, and see what happens when modifying the flash at >12Mhz.  It turns out it may even be a good thing.
http://nerdralph.blogspot.com/2020/09/flashing-avrs-at-high-speed.html

cpldcpu, 2020-09-06:
Interesting!

One challenge I see is that you cannot really measure read margin using the user registers. So, it would be possible that you write the cell successfully, but the charge stored on it is not sufficient to allow for a 10 year data retention. 

Maybe an easy hack could be to write twice? Because it should be fair to assume that the charge stored per time is a constant if all other conditions are the same.

nerdralph, 2020-09-06:
> 
> One challenge I see is that you cannot really measure read margin using the user registers. So, it would be possible that you write the cell successfully, but the charge stored on it is not sufficient to allow for a 10 year data retention.
>

Don't forget charge is 1-(e^-x), so reducing the time by 1/3rd reduces the charge by only 24%.  For retention the t88 spec is actually "20 years at 85°C / 100 years at 25°C".  I suppose I could do an experiment to test retention by writing zeros to the page, then repeatedly verifying the page at high temperature.  If I can find 26-30AWG wire with high temperature insulation, then I could probably do the testing at 150-160C.  However extrapolating the lifetime/temperature curve, I'd need to run the AVR at 200C to get below 1 year expected lifetime.   Given that's above the melting point of my solder, I don't see that experiment happening anytime soon.

However since I've already pushed it two orders of magnitude past it's rated number of write cycles, I may try the repeated read test just to make sure it still holds the zeros for at least a few days.

cpldcpu, 2020-09-06:
>Don't forget charge is 1-(e^-x), so reducing the time by 1/3rd reduces the charge by only 24%

The writing mechanism does most likely not behave like an RC circuit. I don't know what kind of write mechanism is used in the AVR, could be HCI or FN. Usually it is rate limited by the electron transfer process and is operated at strong overbias, so that the gate charge does not really influence the rate.

Maybe it is possible to vary supply voltage to test for programming margin. Since they seem to have an internal charge pump, that may not work, though.

nerdralph, 2020-09-09:
I'm trying to solve a small problem with the new OSCCAL tuner.  At 8Mhz, one USB bit-time is 6.67 cycles.  The counting loop takes 10 cycles, so its possible to miss a single high bit on D+.  That means occasionally it could mistake the EOP SE0 as SOF.
One idea I have is to bump OSCCAL to 255 at the start, and let it tune it down.  With OSCCAL at 255, the clock speed will be over 12Mhz.  A slight downside to this is it would require conditional compilation since we don't want to do it on the tX5 and other devices with a discontinuous OSCCAL curve.
Another idea I have is to add 2 instructions to the loop (in & or) to take 2 samples, each 6 cycles apart.  That guarantees at least one sample per USB bit-time.

Thoughts?

nerdralph, 2020-09-09:
Here's what I've come up with for a double-sampling loop.
```
waitSOF:                                ; 12-cycle loop
    lsr shift
    in scratch, USBIN
    or scratch, r0
    adiw countL, fraction
    bst scratch, USBPLUS
    bld shift, 7
    in r0, USBIN
    sbrs scratch, USBMINUS
    tst shift
    brne waitSOF                        ; not idle last 7 loops
```

cpldcpu, 2020-09-09:
hm... why do you want to run oscall at 8 MHz? There is no V-USB implementation at that frequency?

Maybe another idea to address the issue: How about repeating the fixed point iteration for more cycles and also accumulate fractional count of osccal? Then the error would be smoothed out over time.

nerdralph, 2020-09-09:
> 
> hm... why do you want to run oscall at 8 MHz? There is no V-USB implementation at that frequency?
> 
> Maybe another idea to address the issue: How about repeating the fixed point iteration for more cycles and also accumulate fractional count of osccal? Then the error would be smoothed out over time.

The target speed is not 8MHz, it's 12 on devices like the t88 and t84.  They start off at 8MHz.  The problem is actually more likely than I stated above.  1 low-speed bit(666.7ns) at 12Mhz is 8 cycles, so at 8Mhz it's only 5.333 cycles.

The error from misinterpreting EOP as SOF could be more than an order of magnitude (if the subject EOP is in the 1st 1/10th of the frame).

nerdralph, 2020-09-11:
> Here's what I've come up with for a double-sampling loop.
>
> ```
> waitSOF:                                ; 12-cycle loop
>     lsr shift
>     in scratch, USBIN
>     or scratch, r0
>     adiw countL, [...]

This version fixed the bug with missed D+ pulses, but introduced a problem with frequent missed SOFs.  The problem was checking the USBMINUS bit of the two samples that were or'd together meant that only 3 SE0 samples in a row was guaranteed to be detected.  I changed it to `sbrs r0, USBMINUS` so the check for SE0 is once every loop (12 cycles).  With this change, and with starting OSCCAL with 224 to ensure the initial speed is >12MHz, the t88 I've been testing has been working perfectly.  Dozens of re-enumerations, erase, and upload operations have worked error-free, even using a WRITE_SLEEP time of 3ms and fast-mode.

I'd prefer to avoid the extra code in main to boost OSCCAL when `(OSCCAL_HAVE_XTAL == 0) && (F_CPU == 12000000)`, so I've been thinking about writing another version of the tuner that has a tight 6-cycle loop looking for SE0, and then checks PCIF to see if there was a D+ pulse.  This would allow the tuner to start at 8Mhz and not risk missing the 2-bit SE0 for SOP.

cpldcpu, 2020-09-12:
You could use the pin change interrupt to detect D+ pulses. Then you could use a fast loop to detect SOF in D-. 

btw, one thing that could prove a bit problmeatic with the current loops is a lack of timeout. There were some issues with micronucleus when people had noise on the USB pins even when the device was not connected to the host. In that case sometimes osccall is invoked. This was especially an issues with Olimex devices - no idea why.

nerdralph, 2020-09-19:
After doing some work on usbasp firmware, I'm back to hacking on mn.  One of my ideas is optimizing or even eliminating the CRC16 code.  A couple weeks ago I combined reading descriptors from flash with the CRC function.
https://github.com/nerdralph/micronucleus-firmware/blob/asm/firmware/usbdrv/usbdrvasm.S#L205

I've been thinking that it should be possible to calculate the CRC at compile time.  It would mean breaking up the descriptors into 8-byte chunks, and appending the CRC to each chunk.  That would add 12 bytes to descriptors, and save 42 bytes of the CRC code, for a net gain of 30 bytes.

I've written a small C++ program to test the concept:
```
#include <stdint.h>

int constexpr sum(const uint8_t arr[], int n)
{
    int sum = 0;
    for (int i = 0; i < n; sum += arr[i++]);
    return sum;
}

int main()
{
    uint8_t arr[] = {1, 2, 3};
    return sum(arr, sizeof(arr));
}
```
I tested it with gcc 5.4.0, which needs C++14 support enabled for constexpr functions.
`avr-gcc -std=c++14 -mmcu=avr25 -Os -c const_sum.cpp`

I confirmed that the constexpr function is evaluated at compile time:
```
00000000 <main>:
   0:   86 e0           ldi     r24, 0x06       ; 6
   2:   90 e0           ldi     r25, 0x00       ; 0
   4:   08 95           ret
```

This would have the added benefit of eliminating the runtime cost of calculating the CRC.
Instead of using a constexpr function, another possibility is to write a parser that reads the descriptor definitions, and then generates the segmented descriptors with the CRC added.  The big problem is that although it is technically possible, I can't see a way of implementing this idea without an impact on code readability.

nerdralph, 2020-09-22:
I think I've really gone down the rabbit hole now.  After so many days of reading usb code (both vUSB and usbtiny) I started writing my own.  Here's the plans:
1) really small, probably half the size of vUSB.  I plan to call it picoUSB.
2) 7 cycles/bit so it will work at 10.5MHz, well within the OSCCAL tuning range of 8MHz AVRs
3) to keep it really simple, no bit stuffing (none of the data will have more than 5 sequential "1" bits)  Using device class 0 instead of 0xFF doesn't seem to break anything.
4) if I can figure out a clean way of doing it, compile-time CRC calculation

I already some prototype generating a valid ACK packet (sync, PID, & EOP) in less than 40 asm instructions.  Rx will need a bit more code, probably 50-60 instructions.
It should be slightly more tolerant of timing errors because 7 cycles permits sampling in the center of the bit instead of slightly off-center with 8 cycles per bit at 12MHz.

Here's the core of the tx code:
```
; 7-cycle/bit tx loop, cnt = 2 * # of bits to tx
nextByte:
    breq txEoP
    ld data, X+
tx2bit:
    sbrs data, 0
    out USBPORT-2, mask                 ; even bits
    lpm
    lsr data
    subi cnt, -4
    sbrs data, 0
    out USBPORT-2, mask                 ; odd bits
    brhc nextByte                       ; subi modifies H
    lsr data
    nop
    rjmp tx2bit
```

cpldcpu, 2020-09-23:
That's a cool idea! How do you prevent bit stuffing? By only using one nibble of the payload?

I guess it's a bit more tricky to avoid bit stuffing in the CRC. If it is possible to generate all CRC codes at compile time, one could add a bit of salt, i guess?

nerdralph, 2020-09-23:
Sending inverted nibbles like PIDs is probably the easiest way of sending data.  A 8/6 GCR-like encoding would be preferable if I can come up with a small decoding algorithm.  In either case it would allow for some error checking on the device side.  My reason for wanting more than 4 decoded data bytes per data packet is so I can send R1:R0, Z, and 1 byte for SPMCSR in each user control packet.  It greatly simplifies the bootloader protocol, and is similar to a technique I used in my first picoboot bootloader that made it possible to fit in just 64 bytes.

The CRC isn't so hard since all of the USB responses from the target never change.  If mn had a flash read command for verify, then the CRC problem would be more complicated.  In my mn fork, there's 6 different data packets the device needs to send: 3 for the device descriptor, 2 for the configuration descriptor, and one for the mn config.  Looking at the USB traces, I noticed there was only 1 stuffed bit, and that was for the 0xFF device protocol field.  I think I can get rid of the mn config packet by using the class, subclass, and protocol fields of the device descriptor.  Both the 2nd signature byte and total flash size can be encoded in just 3 bits.  The upper nibble is always 9, and the lower nibble is the flash size in kB as 2^N.  That's why the t88, t84, t85, & m8 all have 0x93 as their middle signature byte.  They all have 2^3 = 8 kB of flash.  Page size only needs a couple bits, and user flash space available could be efficiently encoded as total flash size - bootloader size in pages.

cpldcpu, 2020-09-24:
No need for a complex line coding, I think. You just need to insert two zeroes into each byte, e.g. bit 7 and 0 so that no adjacent bytes will case a long run. Using nibbles may be a bit easier to handle for block transfer, though.

Transferring each byte address together with each cell to be writted is a bit dangerous from a safety perspective, in my opinion, because it is much more difficult to perform a consistency check on client side. It would easily be possible to brick the device. Maybe just transferring the block offset is safer?

[5 more replies]