micronucleus/micronucleus #185: possible enumeration bug

nerdralph, 2020-08-25:
I've started down the rabbit hole of debugging enumeration, using usbmon and wireshark.  One of the errors I've been able to track down relates to the configuration descriptor.  Sometimes (though I haven't been able to deterministically repeat it) Linux will send a get configuration control packet with wLength set to 9.  v-usb will truncate the length of the reply to 9 bytes, but it doesn't modify wTotalLength in the configuration descriptor reply.  This will cause a malformed response error in the Linux USB stack.  I suspect if wTotalLength was 9, the response would be accepted and parsed to see that bNumInterfaces = 1, and it would repeat the get configuration request a second time with wLength = 18.

Rather than adding code to modify wTotalLength on the fly, I removed the interface from the configuration descriptor, set wTotalLength to 9, and bNumInterfaces = 0.  Although unusual, a configuration doesn't have to have any interfaces, since EP0 is associated with the device and not a defined interface.  Both Windows and Linux still properly enumerate, with Linux adding a note in dmesg: "usb 1-1.2: config 1 has no interfaces?"  I've tested this not just with v-usb, but also with a CH552.

Here's the code:
https://github.com/nerdralph/micronucleus-firmware/blob/asm/firmware/usbdrv/usbdrv.c#L99

I'm not suggesting this eliminates enumeration problems, as I still get intermittent errors.  I wonder if part of the problem is related to the polling loop in main not being fast enough to call USB_handler before the sync is almost ended.  I have a couple of USBasp modules that are quite reliable, so I think I'll flash one of them with micronucleus, in order to rule out hardware or oscillator timing problems.

cpldcpu, 2021-03-06:
@nerdralph I am a bit dumbfounded here, to be honest. Is this still an issue in the current code?

nerdralph, 2021-03-06:
I didn't see this after I started using the new OSCCAL tuner, I stopped seeing this problem.
I'm still not sure how a device should be responding to enumeration requests to best comply with the spec, so I can't really say there is any bug here.  Therefore I'll close this issue.

closed 2021-03-06 by nerdralph