micronucleus/micronucleus #186: Releases out-of-date?

ElectricRCAircraftGuy, 2020-08-31:
Are the released hex files out-of-date? 

This, for instance, appears to have installed `firmware version 2.3`, but I expected it to install `firmware version 2.4`:

    micronucleus --run upgrade/releases/upgrade-t85_default.hex

ArminJo, 2020-10-24:
Try [this files](https://github.com/ArminJo/micronucleus-firmware/tree/master/firmware/upgrades)

closed 2020-10-24 by ArminJo

ElectricRCAircraftGuy, 2020-10-24:
@ArminJo, why are we using your fork instead of putting these updates on the original repo? Is your fork the new official repo? If so, why doesn't the original repo state that and direct everyone to your fork?

ArminJo, 2020-10-25:
Good question 👍 
But do not ask me, ask Tim aka cpldcpu

ElectricRCAircraftGuy, 2020-10-25:
@cpldcpu

cpldcpu, 2020-11-18:
@ArminJo 

I think I have stated clearly before, that I prefer to keep Micronucleus neutral. You changed large parts of the documentation in a way that made Micronucleus look like it was associated with Digistump. We discussed this before and I already spent considerable time to dial back your documentation. I still have not received a clear feedback whether you are ok with the way I changed it back. Instead you added a link to your own repository, which I found quite offensive.

This is the reason why I cannot merge your current pull request.

[image]

The part in question is here:

[image]

Frankly, I don't think any OSS projects would include links to forks into their documentation. 

I only have limited time to invest in micronucleus and having to spend it to address this added politicial dimension is really draining my motiviation. There are urgent topics that need to be clarified, like syncing builds for firmware and commandline and ensuring quality control. Also, there are many great other contributions that still need to be merged.

I am definitely open for proposals.

ArminJo, 2020-11-18:
10.7.2020
> Hi Tim,

> you have my OK, but who implements it?
> I had much effort and no result, so I think its up to you now. Take what you think is reasonable, but I am tired to make proposals not beiing accepted.
> BTW. where to put the additional configurations and precompiled hex files?
> 
> Best regards
> Armin

As I stated here, I made a lot of proposals also some without including the "offending" link, but **none** were accepted.
I am sorry, but can not look inside your brain 😞.
Why din't you merge one of the proposals and changed, what you do not like?
As I said I am tired of this discussion, just do what you want (but do something 😉).

BTW. the branch now is totally outdated!

cpldcpu, 2020-11-18:
You have not made a single change to your documentation on your own accord. Your statement that you made proposals is false. Your only proposal was to make a fork, which you very agressively advertised on here. 

This is easily visible from the contribution history of the branch.

Sorry, but I am not willing to have a he-said-she-said discussion like this. So it looks like we are at an impasse. I am very busy in my daily work, so I need to be selective in how I spend time with my hobby project.

cpldcpu, 2020-11-18:
Of course I have to add that your contributions are very valued :). So what do you specifically propose to move forward without me having to comb through your fork and port everything back? I don't think any situation with a fork is acceptable for this project, unless there is a clear technical reason, like e.g. protocol incompatibility.