micronucleus/micronucleus #189: avr/io.h not found

maslaff, 2020-10-09:
Linux. Firmware 2.5 compilation.
Throws an error when do Make or MK_ALL.sh.
Apparently I don't understand something, but I suppose that since there is an include, it means that the library must be somewhere in the system paths or something like that, but I don't have it. Help me figure out what I am missing?

crt1.S:43:20: fatal error: avr/io.h: ...blahblahblah

In general, I want one way or another to add full support for the Digispark board (aka ATtiny85) with Micronucleus 2.5 in PlatformIO. Maybe I don't need to bother with all this compilation, but just fix the existing one? Or maybe someone will show you where to get the finished set.

In any case, I would like to complete what we started.

Sorry for my googleprompt english.

ArminJo, 2020-10-09:
Have you looked at https://github.com/ArminJo/micronucleus-firmware and https://github.com/ArminJo/DigistumpArduino ?

Ho-Ro, 2020-10-09:
> crt1.S:43:20: fatal error: avr/io.h: ...blahblahblah

this *...blahblahblah* tells you probably exactly what's your problem - unfortunately my glass ball is in repair.
Did you search for `avr/io.h`? When I search on my debian stable system (`locate avr/io.h`) it tells me

    /usr/lib/avr/include/avr/io.h

And a quick check where this file comes (`dpkg -S /usr/lib/avr/include/avr/io.h`) tells me that it is installed by package `avr-libc`

    avr-libc: /usr/lib/avr/include/avr/io.h

Did you install the necessary packages?

maslaff, 2020-10-09:
@ArminJo Thanks, i will definitely check it, after I understand better what's what.

@Ho-Ro Yeah! You excellent explainer, man with glass balls. I understood everything even without googleTranslate. I'm install avr-libc, and now this error is gone. 
I didn't copy-paste the output because it's not English. And it became a mess to translate, and so it's clear. Isn't that clear?
Hmm. Translator suggested the word 'clear'. Clear == understand?
And yes, i install necessary packages, about which I managed to find information. Apparently I read it inattentively. Where can I find it?

And so now another problem:
`
collect2: фатальная ошибка: ld завершилась по сигналу 6 [Аварийный останов], core dumped
компиляция прервана.
*** buffer overflow detected ***: terminated
make: *** [Makefile:85: main.bin] Ошибка 1
`

I don’t want to figure out how to do output in English (although it would be worth it to learn the language), so here is a googletranslate translation:
`
collect2: fatal error: ld ended on signal 6 [Emergency stop], core dumped
compilation is interrupted.
***buffer overflow detected ***: terminated
make: *** [Makefile:85: main.bin] Error 1
`

Fuhh! Hard. But incredibly interesting. I'll deal with this morning.
I hope someone will tell me ... Or send me to create a new issue.

Ho-Ro, 2020-10-09:
> I don’t want to figure out how to do output in English 

Just type e.g. `LANG=C make` to switch the output of "make" to English.

maslaff, 2020-10-09:
@Ho-Ro Ohh. Now, with LANG =, it outputs:
```

$ LANG=C make
upgrade.c: In function 'write_new_bootloader':
upgrade.c:94:23: error: 'bootloader_size' undeclared (first use in this function)
   while ( page_addr < bootloader_size ) {
                       ^
upgrade.c:94:23: note: each undeclared identifier is reported only once for each function it appears in
upgrade.c:98:32: error: 'bootloader' undeclared (first use in this function)
       int subaddress = ( (int) bootloader ) + page_addr + word_addr;
                                ^
upgrade.c:99:33: error: 'bootloader_end' undeclared (first use in this function)
       if ( subaddress < ( (int) bootloader_end ) ) { // valid boot code
                                 ^
upgrade.c:107:17: error: 'bootloader_address' undeclared (first use in this function)
     erase_page( bootloader_address + page_addr );
                 ^
upgrade.c: In function 'forward_interrupt_vector_table':
upgrade.c:153:39: error: 'bootloader_address' undeclared (first use in this function)
     vector_table[ iter ] = 0xC000 + ( bootloader_address / 2 ) - 1;
                                       ^
make: *** [Makefile:117: upgrade.o] Error 1

```

tomaskovacik, 2020-10-10:
read note in upgrade.c:

// Upgrade is an in-place firmware upgrader for tiny85 chips - just fill in the
// 'bootloaderAddress' variable in bootloader_data.h, and the bootloaderData
// progmem array with the bootloader data, and you're ready to go.

ArminJo, 2020-10-10:
@tomaskovacik Thanks, I will check tomorrow

Ho-Ro, 2020-10-10:
Hi, if you're on Linux change the Makefile accordingly (the github version is probably for Windows):

    --- Makefile~   2020-10-10 11:17:10.703467201 +0200
    +++ Makefile    2020-10-10 11:25:01.574037466 +0200
    @@ -151,15 +151,15 @@
     bootloader.h: bootloader.o
     #      @echo generating bootloader.h
     # for internal echo command
    -       @echo extern const uint8_t $* [] PROGMEM;> $@
    -       @echo extern const uint8_t $*_end [] PROGMEM;>> $@
    -       @echo extern const uint8_t $*_size_sym [];>> $@
    -       @echo #define $*_size ( (int) $*_size_sym )>> $@
    -       @echo #define $*_address 0x$(BOOTLOADER_ADDRESS)>> $@
    +#      @echo extern const uint8_t $* [] PROGMEM;> $@
    +#      @echo extern const uint8_t $*_end [] PROGMEM;>> $@
    +#      @echo extern const uint8_t $*_size_sym [];>> $@
    +#      @echo #define $*_size ( (int) $*_size_sym )>> $@
    +#      @echo #define $*_address 0x$(BOOTLOADER_ADDRESS)>> $@
 
     # for unix echo command
    -#      @echo "extern const uint8_t" $*"[] PROGMEM;" > $@
    -#      @echo "extern const uint8_t" $*_end"[] PROGMEM;" >> $@
    -#      @echo "extern const uint8_t" $*_size_sym"[];" >> $@
    -#      @echo "#define $*_size ( (int) $*_size_sym )" >> $@
    -#      @echo "#define $*_address 0x$(BOOTLOADER_ADDRESS)" >> $@
    +       @echo "extern const uint8_t" $*"[] PROGMEM;" > $@
    +       @echo "extern const uint8_t" $*_end"[] PROGMEM;" >> $@
    +       @echo "extern const uint8_t" $*_size_sym"[];" >> $@
    +       @echo "#define $*_size ( (int) $*_size_sym )" >> $@
    +       @echo "#define $*_address 0x$(BOOTLOADER_ADDRESS)" >> $@

ArminJo, 2020-10-12:
Thanks, I improved the upgrade documentation to cover the `'bootloader_size' undeclared` error.

closed 2020-10-12 by ArminJo

tomaskovacik, 2020-10-12:
@ArminJo you still have compile instraction pointing to https://github.com/ArminJo/micronucleus-firmware/tree/master/firmware#compiling just pointing out :) 

also, if there is all text already in makefile, why not simply "if it" ? https://stackoverflow.com/questions/714100/os-detecting-makefile

just sayin :) this works for me on ubu:

diff --git a/firmware/Makefile b/firmware/Makefile
index da5d324..eed4991 100644
--- a/firmware/Makefile
+++ b/firmware/Makefile
@@ -151,15 +151,17 @@ bootloader.o: bootloader.raw
 bootloader.h: bootloader.o
 #      @echo generating bootloader.h
 # for internal echo command
+ifeq ($(OS),Windows_NT)
        @echo extern const uint8_t $* [] PROGMEM;> $@
        @echo extern const uint8_t $*_end [] PROGMEM;>> $@
        @echo extern const uint8_t $*_size_sym [];>> $@
        @echo #define $*_size ( (int) $*_size_sym )>> $@
        @echo #define $*_address 0x$(BOOTLOADER_ADDRESS)>> $@
-
+else
 # for unix echo command
-#      @echo "extern const uint8_t" $*"[] PROGMEM;" > $@
-#      @echo "extern const uint8_t" $*_end"[] PROGMEM;" >> $@
-#      @echo "extern const uint8_t" $*_size_sym"[];" >> $@
-#      @echo "#define $*_size ( (int) $*_size_sym )" >> $@
-#      @echo "#define $*_address 0x$(BOOTLOADER_ADDRESS)" >> $@
+       @echo "extern const uint8_t" $*"[] PROGMEM;" > $@
+       @echo "extern const uint8_t" $*_end"[] PROGMEM;" >> $@
+       @echo "extern const uint8_t" $*_size_sym"[];" >> $@
+       @echo "#define $*_size ( (int) $*_size_sym )" >> $@
+       @echo "#define $*_address 0x$(BOOTLOADER_ADDRESS)" >> $@
+endif

Ho-Ro, 2020-10-12:
>@ArminJo you still have compile instruction pointing to https://github.com/ArminJo/micronucleus-firmware/tree/master/firmware#compiling ,just pointing out :)

Please change this line in top level README.md from 

    # Compile instructions for the bootloader are [here](firmware#compiling)

to

    # Compile instructions for the bootloader are [here](firmware/upgrades/README.md#prerequisits)

and add this line to the `firmware/upgrades/README.md#prerequisits` section to guide the Linux users:

    For Linux install the compiler, library and tools with your package system, e.g. for debian use `apt install gcc-avr avr-libc, binutils-avr`.

Especially Arduino users may have the libraries already installed somewhere, but not in the correct path.

Thank you for maintaining this nice bootloader - I use VUSB since >10 years and I see it still as an outstanding engineering solution.

ArminJo, 2020-10-12:
@tomaskovacik Thanks a lot, I modified the makefile, but depending on the toolchain, you require the unix echo on windows too. Thats the reason, I was reluctant to do the change earlier. And I found the missing Readme 😀 .
@Ho-Ro Thank you too, I included your changes (and modified it slightly).