micronucleus/micronucleus #190: Micronucleus support in AVRDUDE

mariusgreuel, 2020-10-18:
I added experimental support for the micronucleus bootloader to AVRDUDE.

I would appreciate if any interested party could test these patches and provide feedback. Please see the documentation in readme.md about the command-line usage.

There are pre-compiled binaries for Windows, and also a Linux branch that just contains the patches for micronucleus (which you need to build yourself).

https://github.com/mariusgreuel/avrdude

ArminJo, 2021-01-04:
Hi Marius
I tested your new avrdude.
I was able to upload hex files. 👍 
But I can not start avrdude and then attach the Digispark.

BTW. the id = "linuxspi"; programmer is missing, so I have to edit my existing config file, after adding your micronucleus programmer.
-

ArminJo, 2021-02-04:
For windows it works now. Thanks!

closed 2021-02-04 by ArminJo