micronucleus/micronucleus #191: WinUSB support for micronucleus

mariusgreuel, 2020-10-18:
Some time ago, I created a fork of the micronucleus repo, to add WinUSB support. Essentially, it is a patch to the bootloader that adds an extra Microsoft OS description that enables automatic load of the WinUSB driver. Note that the WinUSB driver requires the command-line tool to support it, such as https://github.com/mariusgreuel/avrdude.
 
Is there any interest in integrating this change into your repo? I realize this adds some extra bytes to the bootloader, and it is of no use to the Linux community. However, I think there are a number of reasons to get rid of the libusb-win32 driver for Windows and switch to the native Windows driver.

See my fork at
https://github.com/mariusgreuel/micronucleus

cpldcpu, 2020-10-19:
That definitely looks very interesting. Could the integration into avrdude also be made backwards compatible, so that also old devices can be flashed?

I am not too familiar with WinUSB. This would require changing the device descriptor in a way the is not compatible to the libusb setup, right? So basically a protocol update would be necessary.

ArminJo, 2021-01-04:
Seems this increases the firmware size too much.
One part, only for example:
```
//#define USB_CFG_DEVICE_NAME 0x00B5,'B'
//#define USB_CFG_DEVICE_NAME_LEN 2
#define USB_CFG_DEVICE_NAME 'M', 'i', 'c', 'r', 'o', 'n', 'u', 'c', 'l', 'e', 'u', 's', ' ', 'B', 'o', 'o', 't', 'l', 'o', 'a', 'd', 'e', 'r'
#define USB_CFG_DEVICE_NAME_LEN 23
```
Thanks, but I doubt anyone will install the bigger firmware.

closed 2021-01-04 by ArminJo