micronucleus/micronucleus #192: Avoid the first page of flash?

ericdraken, 2020-11-26:
How can micronucleus source code or config headers be modified so user code starts uploads to and starts from the **second** page of memory (or even the third)? How to set the start offset to n*PAGE_SIZE? Why? Because the first page of memory is sometimes corrupted like so:

[image]

cpldcpu, 2020-11-26:
When does this happen? Which version of Micronucleus are you using?

Writing to the first page of flash cannot be omitted, since it contains the the reset vector that is needed to enter the bootloader.

ericdraken, 2020-11-26:
This happens when the power cycles and the voltage drops slowly. It happens **sometimes**, but it does happen. BOD is a good stop-gap measure, but it isn't 100% effective.

When I dump the factory flash of a new ATTiny85, I see

[image]

and the bootloader starts:

[image]

It looks like the bootloader doesn't need a jump from the start of memory. So, it feels like my request is feasible.

cpldcpu, 2020-11-26:
Can you check which version of micronucleus are you using?

0xffff is interpreted as "NOP" by AVR, so if all blocks before the bootloader are erased, the bootloader will be started. This will not work if Bank0 is erased and there is something else in Bank1. A jmp instruction is required in Bank0 to enter the bootloader.

ericdraken, 2020-11-27:
Micronucleus 2.04.

I have two possible solutions: 

1) I can compile the application code to ihex and store that in an array in the bootloader itself, then compile the bootloader and flash that. Then I can use the NOP sled from address 0 to the fat bootloader and voila. 

2) I can edit the bootloader (somehow?) to tell it that application flash starts at page 2. Since BOOTRST=0 and the reset vector points to the bootloader, if I modify the bootloader, I can start the application code at any point I wish. Then, what do I modify in the bootloader to start at the second page?

cpldcpu, 2020-11-28:
I still don't understand why you want to address fixing the symptons only. This kind of failure you are describing is something that should specifically not happen in micronucleus. The programming algorithm first erases the entire use space starting from the last bank going down to 0. The it begins writing at bank 0 upwards. This ensures that either a nop-sled is in place or a valid rjmp exists in bank 0.

So you either stumbled upon a previously unknown bug or there is something else that erases page 0? Is there anything in your user program that could invoke erasing of bank 0?

Are you using ENABLE_UNSAFE_OPTIMIZATIONS? 

In any case, if you really wanto to fix the symptons only, you could put an rjmp to the bootloader as first instruction in bank 1.

closed 2021-01-04 by ArminJo