micronucleus/micronucleus #194: How to re-flash the bootloader?

dentex, 2020-12-07:
I have a cheap digispark clone, the one with the USB plug.
I think that I messed up the bootloader.
How to re-flash it?

At the moment the board (it seems) is not recognized when plugged in.

`dmesg` reports only:
```[ 5711.716266] usb 3-2: new low-speed USB device number 76 using xhci_hcd
[ 5711.848246] usb 3-2: device descriptor read/64, error -71
[ 5712.084255] usb 3-2: device descriptor read/64, error -71
[ 5712.320236] usb 3-2: new low-speed USB device number 77 using xhci_hcd
[ 5712.448242] usb 3-2: device descriptor read/64, error -71
[ 5712.684306] usb 3-2: device descriptor read/64, error -71
[ 5712.792295] usb usb3-port2: attempt power cycle
[ 5713.444268] usb 3-2: new low-speed USB device number 78 using xhci_hcd
[ 5713.444846] usb 3-2: Device not responding to setup address.
[ 5713.652844] usb 3-2: Device not responding to setup address.
[ 5713.860180] usb 3-2: device not accepting address 78, error -71
[ 5713.992164] usb 3-2: new low-speed USB device number 79 using xhci_hcd
[ 5713.992608] usb 3-2: Device not responding to setup address.
[ 5714.200816] usb 3-2: Device not responding to setup address.
[ 5714.408251] usb 3-2: device not accepting address 79, error -71
[ 5714.408377] usb usb3-port2: unable to enumerate USB device
```

while a working one (with a simple "blink" script running) shows:
```[ 5727.808106] usb 3-2: new low-speed USB device number 80 using xhci_hcd
[ 5727.958454] usb 3-2: New USB device found, idVendor=16d0, idProduct=0753, bcdDevice= 2.02
[ 5727.958457] usb 3-2: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 5733.539786] usb 3-2: USB disconnect, device number 80
   ---  [ the below section seems identical, and appear when the device starts executing the script] ---
[ 5734.107997] usb 3-2: new low-speed USB device number 81 using xhci_hcd
[ 5734.240003] usb 3-2: device descriptor read/64, error -71
[ 5734.475933] usb 3-2: device descriptor read/64, error -71
[ 5734.712024] usb 3-2: new low-speed USB device number 82 using xhci_hcd
[ 5734.840106] usb 3-2: device descriptor read/64, error -71
[ 5735.075948] usb 3-2: device descriptor read/64, error -71
[ 5735.184065] usb usb3-port2: attempt power cycle
[ 5735.836015] usb 3-2: new low-speed USB device number 83 using xhci_hcd
[ 5735.836513] usb 3-2: Device not responding to setup address.
[ 5736.044384] usb 3-2: Device not responding to setup address.
[ 5736.252010] usb 3-2: device not accepting address 83, error -71
[ 5736.379969] usb 3-2: new low-speed USB device number 84 using xhci_hcd
[ 5736.380462] usb 3-2: Device not responding to setup address.
[ 5736.588489] usb 3-2: Device not responding to setup address.
[ 5736.796004] usb 3-2: device not accepting address 84, error -71
[ 5736.796125] usb usb3-port2: unable to enumerate USB device
```

dentex, 2020-12-07:
Oh maybe I get it now... I messed up the V-USB code and so, for re-programming, I need an external device working as ISP, as mentioned here? https://makersportal.com/blog/2018/5/19/attiny85-arduino-board-how-to-flash-the-arduino-bootloader-and-run-a-simple-sketch

ArminJo, 2020-12-07:
To burn micronucleus bootloader on a chip use the `Burn_avrdude-*` scripts available [here](https://github.com/ArminJo/micronucleus-firmware/tree/master/utils).

dentex, 2020-12-07:
I already tried something similar using `avrdude`, but the problem seems to be the usb device not recognized. I'll give it a try anyway (although I'm on a Linux machine, but I think I can adapt the commands).

dentex, 2020-12-08:
Board recovered! I used indications I found here: https://arduino.stackexchange.com/a/44704
Working good now.
Thanks everyone.

closed 2020-12-08 by dentex