micronucleus/micronucleus #195: Code is not getting uploaded to Digispark attiny85

Tirumalesh8686, 2020-12-28:
I am using Digispark Attiny85 to work on a project.

I am following the below link to use DigiSpark Attiny85.
https://www.instructables.com/Digispark-Attiny-85-With-Arduino-IDE/
 
At the end, I am not getting "Micro nucleus Done, Thankyou..!! "

Instead it is eventually showing:
Running Digispark Uploader...
Plug in device now... (will timeout in 60 seconds)
What I have noticed is that the status bar goes from Uploading... to Done uploading. in less than a second, but nothing is uploaded eventually.
Did you notice such a similar behavior?
Did you find a way to solve this issue?

ArminJo, 2020-12-28:
To avoid errors, use the new boardmanager URL https://raw.githubusercontent.com/ArminJo/DigistumpArduino/master/package_digistump_index.json and install the Digistump AVR Boards version 1.6.8 or later.

blippy, 2021-01-02:
I tried the URL you suggested. I'm getting

```
Device reports version as: 2.2
Warning: device with unknown new version of Micronucleus detected.
This tool doesn't know how to upload to this new device. Updates may be available.
```

lsusb reports

```
Bus 005 Device 005: ID 16d0:0753 MCS Digistump DigiSpark
```

closed 2021-01-26 by ArminJo