micronucleus/micronucleus #196: Digispark not recognized on Linux machine, but only at boot.

dentex, 2020-12-31:
Hello.
I have a problem in getting the board detected by my Linux machine if it's already connected at boot time.
The only message I see with `dmesg` is:
```
[    1.500000] usb 3-2: new low-speed USB device number 3 using xhci_hcd
[    1.600000] usb 3-2: device descriptor read/64, error -71
```
...and nothing works. The device remains unresponsive (and the led is fixed ON).

If the device is plugged **after** the machine booted, all goes well:
```
[   70.858428] usb 3-2: new low-speed USB device number 10 using xhci_hcd
[   71.008647] usb 3-2: New USB device found, idVendor=16d0, idProduct=0753, bcdDevice= 2.05
[   71.008649] usb 3-2: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[   76.921597] usb 3-2: USB disconnect, device number 10
```
and then after 6 seconds:
```
[   77.441563] usb 3-2: new low-speed USB device number 11 using xhci_hcd
[   77.901798] usb 3-2: New USB device found, idVendor=16c0, idProduct=27db, bcdDevice= 1.00
[   77.901805] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   77.901808] usb 3-2: Product: DigiKey
[   77.901811] usb 3-2: Manufacturer: digistump.com
[   78.022308] input: digistump.com DigiKey as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/0003:16C0:27DB.0003/input/input28
[   78.082004] hid-generic 0003:16C0:27DB.0003: input,hidraw2: USB HID v1.01 Keyboard [digistump.com DigiKey] on usb-0000:00:14.0-2/input0
```

I tried various `micronucleus` versions with different boards (the ones with the micro USB connectors and the classical ones), also the fork by @ArminJo 

Do you have any suggestion in regards?
Thanks.

ArminJo, 2021-01-04:
This is the expected behavior 😀 
The device waits for 6 seconds in the bootloader and then starts its application, which normally means a disconnect from USB.

dentex, 2021-01-05:
I understand the intended behavior about "wait 6 seconds" and then "run the user program". The problem during boot is that the device remains unresponsive. It doesn't wait in the user-program state. It actually doesn't get recognized at all. So I'm not sure that it should work like this: basically it needs to be plugged in the PC's USB port after boot, invariably.

ArminJo, 2021-01-05:
This sound more as a problem of linux and its USB driver than a problem of micronucleus or do I miss something?

dentex, 2021-01-05:
I honestly don't know.

ArminJo, 2021-01-05:
So do I

Ho-Ro, 2021-01-05:
I did a quick check with a debian system:
`5.9.0-0.bpo.2-amd64 #1 SMP Debian 5.9.6-1~bpo10+1 (2020-11-19) x86_64 GNU/Linux` on my Thinkpad T400 with a digispark board with setup [t85_entry_on_powerOn_fastExit](https://github.com/ArminJo/micronucleus-firmware/tree/master/firmware/configuration/t85_entry_on_powerOn_fastExit) and this is what `dmesg` says when the computer boots with digispark plugged in:
````
[    2.823723] usb 6-2: New USB device found, idVendor=16c0, idProduct=27db, bcdDevice= 1.00
[    2.823725] usb 6-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.823727] usb 6-2: Product: EasyLogger
[    2.823728] usb 6-2: Manufacturer: obdev.at
...
[    2.857178] input: obdev.at EasyLogger as /devices/pci0000:00/0000:00:1d.0/usb6/6-2/6-2:1.0/0003:16C0:27DB.0002/input/input5
````

dentex, 2021-01-05:
I had the problem with both stock firmware and @ArminJo  _t85_entry_on_powerOn_activePullup_fastExit_
(recommended configuration).
@Ho-Ro Do you think is worth trying your config?

Ho-Ro, 2021-01-05:
If I would like to fix my problem then I would try a config that was reported as working :)

dentex, 2021-01-06:
I tried the *t85_entry_on_powerOn_fastExit* config, but it does the same.
It is sufficient to wait a few seconds to plug the device in, as soon as the boot starts, and it works.
Very strange.

I tried using an USB port that I have and that stays powered also while the laptop is off: same result (onboard LED flashes and then stays ON; the digispark stays unresponsive until power a cycle). Maybe it depends on some sort of "wrong" or missing answer via USB? I don't know.

Just to clarify, I have the same result on another Laptop (with the same Linux Mint 20 version) and on other digispark boards I had around.

closed 2021-02-04 by ArminJo

PizzaProgram, 2021-02-04:
Closing the issue does not make this problem "go away" ... 
It is a known bug, that if a host's USB port stays powered while rebooting, (like at a Raspberry Pi), micronucleous hangs and never works again, unless unplugged + replugged.

dentex, 2021-02-04:
> It is a known bug
Hi. Do you have any link or other source of information to look at?

cpldcpu, 2021-02-04:
It sounds like for some reason the bus reset is not detected and the device does not enumerate properly.

ArminJo, 2021-02-04:
May I conclude...
You want to have Micronucleus active even after a reset. -> micronucleus active means: waiting for the host to send a new program. -> Your user program should not run after a reset.
What is the usecase of this behavior?