micronucleus/micronucleus #197: Cannot build upgrade on ubuntu 20.04

GhostDog98, 2021-01-16:
Ruby Version: ruby 2.7.0p0 (2019-12-25 revision 647ee6f091) [x86_64-linux-gnu]

Following the instructions inside of "technical details.txt" (located in /master/upgrade):
First i run "ruby ./generate-data.rb ./hexfilehere.hex"
Which results in bootloader_data.c being created, which seems to be intended, then, as per the instructions, i run "make clean; make" (i have make 4.2.1-1.2), which results in the following error:

`rm -f upgrade.hex upgrade.bin *.o upgrade.s upgrade.map upgrade.c.lst`
`avr-gcc -Wall -Os -fno-move-loop-invariants -fno-tree-scev-cprop -fno-inline-small-functions -I. -Ilibs-device -mmcu=attiny85 -DF_CPU=16500000 -DAPP_ADDRESS=0x80  -c upgrade.c -o upgrade.o -Wa,-ahls=upgrade.c.lst`
`In file included from upgrade.c:21:0:`
`./utils.h:1:30: fatal error: util/delay_basic.h: No such file or directory`
`compilation terminated.`
`make: *** [Makefile:156: upgrade.o] Error 1`

MikhailKoslowski, 2021-01-23:
I have the same Ubuntu as yours and was able to build everything without any problem.
It seems that the missing file is from the avr toolchain (util/delay_basic.h). I would try reinstalling it first.
`sudo apt-get install gcc-avr avr-libc avrdude`

you probably don't need the avrdude package if you are not flashing using the ISP, and the file you need should be in the avr-libc package.

GhostDog98, 2021-01-25:
Yup, uploaded fine, but now it's not being detected, although i suspect that's another different problem...

MikhailKoslowski, 2021-01-25:
I'm glad it worked.
I had a similar problem in my last update, after uploading it my digispark was not detected again.
mine had the reset pin enabled so I was able to flash it again using avrdude with the buspirate.
If your board have the reset pin enabled you should be able to flash it again using an Arduino or any other avr programming tool. Otherwise look for a HV programmer and you should be able to get it working again.

closed 2021-01-25 by ArminJo