micronucleus/micronucleus #205: Compile support for attiny2313/4313

MartinHerren, 2021-01-28:
Would supporting the attiny2313/4313 be possible or does it suffer from the same limitations as the 1614 ?

No problem if it is not possible. If it is only not supported because nobody did it yet, i'd try to add it if there is no known limitation.

cpldcpu, 2021-01-31:
The 2313 does only have 2kb of flash? I don't think there is a point to using micronucleus on that device. 4314 may be marginal.

In principle it could work. Not sure how stable the RC-oscillator on the devices is, maybe you need an external XTAL.

MartinHerren, 2021-01-31:
Thanks for your quick response.

Yes, it doesn't make much sense for the 2313 but in this case i'll try to test with a 4313 if there doesn't seem a reason it couldn't be supported. And yes it will require an external XTAL as the internal clock source is only 8MHz which isn't supported by v-usb.

closed 2021-03-06 by cpldcpu

RobertDaleSmith, 2021-07-04:
@MartinHerren Were you ever able to get this working with 4313? I got mine to compile but having trouble with USB being recognized properly after flashing.