micronucleus/micronucleus #210: Do not build executables, if e.g. Readme has changed.

ArminJo, 2021-02-02:
@quinot  Please modify the main.yml accordingly, to build binaries only if one of the files in commandline has changed.
It confuses all users, if we create new binaries **with no changes**.

cpldcpu, 2021-02-02:
But actually also a new zip file of the entire repo is generated. It makes a lot of sense to do that, even if only the readme was changed.

It's pretty normal for open source projects to have a continuous-build and a release build. We need to tag the current repo to generate the 2.5 release.

ArminJo, 2021-02-02:
> But actually also a new zip file of the entire repo is generated. It makes a lot of sense to do that, even if only the readme was changed.

I totally agree! But I do not like the idea of building new exectutables just for fun, without any reason!

> Now that we have automatically generated releases as archive we can put anything there that we want to have. There is no need to have executables in the repository - eventually we could also automate the hex file build and put them in a separate archive. I think that's also much easier for users who are not looking into compiling everything themselves.

I agree too! But first let us have a concept, a binaries and bootloader repo and a workflow script for the bootloader files.

cpldcpu, 2021-02-02:
I have though a bit about this. Doing a proper conditional build is quite hard, since you would have to make sure to check all dependencies. Maybe the easiest solution would simple be to switch the build script to do nightly builds instead doing a build for every commit? That's how most of the other projects are handling it, i think. We would have to disable the pushback for the nightly built though, because otherwise things get really messy.

ArminJo, 2021-02-02:
> Doing a proper conditional build is quite hard,

It is as easy as could be 😀  see https://github.com/micronucleus/micronucleus/blob/b6cea26a00d56e0ca1d1266e4159c8e9c8df99e3/.github/workflows/BuildBinaries.yml Line 10ff

cpldcpu, 2021-02-02:
I guess you are right :)

ArminJo, 2021-02-04:
@quinot 
Hi Thomas, 
are you still developing your fantastic script, or may I apply a few changes to it?
Regards
Armin

quinot, 2021-02-04:
@ArminJo I'll be merging support for ARM binaries in the coming days (sorry, can't commit to a more precise time frame).
If you'd like to make detail adjustments please go ahead and I'll rebase.
If you're considering major reorganization I'd be happy to participate to discussion if you think I can be of some help.
Either way, don't hesitate to tag me in pull requests.

ArminJo, 2021-02-04:
Thanks for the quick response.
I am currently packaging the micronucleus for a Arduino core and discovered that there are [standard names](https://arduino.github.io/arduino-cli/package_index_json-specification/) used for the different OS. I changed the [directory names of the builds](https://github.com/micronucleus/micronucleus/tree/master/commandline/builds) accordingly. 
It would be great, if you can use the same naming convention for your zips. e.g. micronucleus-2.5-x86_64-apple-darwin.zip or micronucleus-cli-2.5-arm-linux-gnueabihf.zip or micronucleus-cli-2.5-i686-mingw32.zip.
Thanks
Armin

closed 2021-02-09 by ArminJo