micronucleus/micronucleus #212: [BUG] custom compilation fails in WSL

DrMullings, 2021-02-04:
**To Reproduce**
Steps to reproduce the behavior:
1. cloned repository
2. copied t85_default to t85_custom
3. `#define ENTRYMODE ENTRY_JUMPER` in header
4. `make CONFIG=t85_custom`
=> 
```
/usr/lib64/gcc/avr/7/ld: warning: -z relro ignored
/usr/lib64/gcc/avr/7/ld: main.bin section `.note.gnu.build-id' will not fit in region `text'
/usr/lib64/gcc/avr/7/ld: region `text' overflowed by 28 bytes
collect2: error: ld returned 1 exit status
make: *** [Makefile:81: main.bin] Error 1
```

**Expected behavior**
It would be nice if it compiled

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Development environment: WSL openSUSE Leap 15.1
 - Version of the bootloader: nA

cpldcpu, 2021-02-04:
Which version of avr-gcc are you using? In any way, it appears that your version of AVR-GCC generates a binary that is too large.
Short term fix would be to change the address of the bootloard in the confiuration file to a lower one. See comments in the makefile.inc.

closed 2021-02-04 by cpldcpu

DrMullings, 2021-02-04:
I am using:
avr-gcc (SUSE Linux) 7.4.0
Which version is suggested?
The documentation on how to calulate this is a pain in the butt and the python script wont work if it does not compile

ArminJo, 2021-02-04:
Set `BOOTLOADER_ADDRESS = 19C0` in Makefile.inc

DrMullings, 2021-02-04:
> 
> 
> Set `BOOTLOADER_ADDRESS = 19C0` in Makefile.inc

That is already set by default

cpldcpu, 2021-02-04:
set it to a low value, e.g. 1800, compile and the update with the output of the python script.

DrMullings, 2021-02-04:
That worked, thanks a lot!
Should I still take another gcc version to save space? If yes which version?

cpldcpu, 2021-02-04:
the current one :)

DrMullings, 2021-02-04:
Thanks for the quick and good support