micronucleus/micronucleus #217: Compile error with -static on Arch Linux

yjun123, 2021-02-28:
Please also have a look at discussions: https://github.com/micronucleus/micronucleus/discussions

**Describe the bug**
-static in Makefile cause compile error since Arch Linux doesn't provide static library in package libusb-compat 0.1.7,
but after remove -static , compile with dynamic libaray, it works. 

**To Reproduce**
execute make in commandline dir,

> Building library: library/micronucleus_lib.c...
> gcc  -Ilibrary -O -g -D LINUX -c library/micronucleus_lib.c
> Building library: library/littleWire_util.c...
> gcc  -Ilibrary -O -g -D LINUX -c library/littleWire_util.c
> Building command line tool: micronucleus...
> gcc  -Ilibrary -O -g -D LINUX -o micronucleus micronucleus.c micronucleus_lib.o littleWire_util.o -static -L/usr/lib -lusb
> /usr/bin/ld: cannot find -lusb
> collect2: error: ld returned 1 exit status
> make: *** [Makefile:59: micronucleus] Error 1

**Expected behavior**
I want know why set static complilation in Makefile instead of dynamic compilation.
is It necessary.

<https://github.com/micronucleus/micronucleus/commit/44a7664bd4c86b86596fe13bcdae7428fe1c4907#diff-460dfaa0de8e1a00b5454e731bfc7c02142714388df899827f02ca2dc81d8469>

**Used Hardware**
Add information about the board you have been using here

**Desktop (please complete the following information):**
- OS: Arch Linux with kernel 5.10.17-1-lts
- libusb 1.0.24
- libusb-compat 0.1.7
- gcc 10.2.0

cpldcpu, 2021-02-28:
This seems to be related to this pull request, if I am not mistaken: https://github.com/micronucleus/micronucleus/pull/208 ?

@stonehippo any idea?

stonehippo, 2021-02-28:
@cpldcpu @yjun123 I made the compilation static so that the micronucleus CLI would be portable. The previous dynamic build required the install of libusb/libusb-compat separately, which made install a bit harder when used for tooling (like with ATTinyCore or DigiStump, which bundle the binary). I originally did this in my own repo, but the changes got picked up, even though they had a bug, hence #208.

The original Makefile assumed a dynamic build, with the option to edit it for a static build, though the config was a bit off, which is what my changes addressed. My assumption (which I should have noted), was that anyone building the CLI for themselves would have the libs installed or could install them, with the static binary as the preferred default case. 

That said, I can make a PR to the Makefile to allow for static compilation as the default, and dynamic via a flag, if that's desired.

stonehippo, 2021-02-28:
@yjun123 would you mind letting me know which packages you have installed to get Arch Linux to compile the CLI? I've got a test install up is iSH, and so far I've thrown in:

- gcc 
- make
- bash 
- gif
- libusb
- libusb-compat
- libusb-dev
- libusb-compat-dev

I don't use Arch all the time, so I'll need a little guidance on how you're set up.

cpldcpu, 2021-03-05:
>That said, I can make a PR to the Makefile to allow for static compilation as the default, and dynamic via a flag, if that's desired.

That sounds like a good solution. Although people still need to figure out that the flag will allow to fix compilation errors if no static library is present.

cpldcpu, 2021-03-07:
Well, the binaries are build using github actions anyways.

I took a pragmatic approach and added a comment to the makefile:
https://github.com/micronucleus/micronucleus/pull/218/commits/be041cd43926de0ad9080fd89a24726cd50e3f15#

closed 2021-03-07 by cpldcpu