micronucleus/micronucleus #219: [BUG]

nerdralph, 2021-02-28:
This causes builds to fail in mingW:
https://github.com/micronucleus/micronucleus/blob/master/firmware/Makefile#L171
The reason is $OS matches Windows_NT, but the shell is bash, not CMD.exe.

Is the Windows code really needed?  Builds in mingW work fine without the Windows_NT special case.

cpldcpu, 2021-03-05:
hm... this is needed if you want to build the firmware with the atmel toolchain, for example. (Atmtel studio command prompt).  Need to figure out a way to resolve that.

cpldcpu, 2021-03-05:
One could probably detect the shell type `SHELL=/bin/sh` it appears this is set in the MINGW-shell, but not in the windows shell.

nerdralph, 2021-03-05:
> 
> 
> hm... this is needed if you want to build the firmware with the atmel toolchain, for example. (Atmtel studio command prompt). Need to figure out a way to resolve that.

OK, you're on your own there.  Atmel studio is a pig, and isn't worth the space on my hard drive.

cpldcpu, 2021-03-06:
Oh come on, religious discussions about toolchains don't lead to anywhere. It's clear that the Windows shell comes with Windows and hence is likely to be used as build environment, regardless of where the compiler-binares are coming from. (Another source would be the Arduino distribution).

cpldcpu, 2021-03-06:
@nerdralph I submitted an updated makefile to your pull request. Can you check whether it works?

closed 2021-03-06 by cpldcpu

nerdralph, 2021-03-06:
> 
> 
> @nerdralph I submitted an updated makefile to your pull request. Can you check whether it works?

OK on mingw32, but I think Mingw64(MSYS2) will still have the problem because by default $SHELL == /bin/bash, not /bin/sh.

nerdralph, 2021-03-06:
> 
> 
> > @nerdralph I submitted an updated makefile to your pull request. Can you check whether it works?
> 
> OK on mingw32, but I think Mingw64(MSYS2) will still have the problem because by default $SHELL == /bin/bash, not /bin/sh.

It surprisingly works fine.  I didn't realize gnu make doesn't use the SHELL env var, it has it's own system for setting SHELL.
https://www.gnu.org/software/make/manual/html_node/Choosing-the-Shell.html