micronucleus/micronucleus #223: [BUG] register use still not fixed

nerdralph, 2021-03-08:
Reported here before:
https://github.com/ArminJo/micronucleus-firmware/issues/11

But the issue was closed without fixing.  The comment here looks like the issue was not understood:
https://github.com/micronucleus/micronucleus/blob/master/firmware/usbdrv/usbdrvasm165.inc#L41

Despite the comment, avr-gcc does rely on r1[67] and Y being callee-saved.  Here's one of many examples that can be found in MN:
```
    1eda:       8c 81           ldd     r24, Y+4        ; 0x04
    1edc:       9d 81           ldd     r25, Y+5        ; 0x05
    1ede:       c6 df           rcall   .-116           ; 0x1e6c <writeWordToPageBuffer>
```
It's just a lucky fluke that avr-gcc didn't generate code that saves anything important in Y before calling USB_handler.

In my asm branch, I fixed the 12, 16, & 16.5 versions.
https://github.com/nerdralph/micronucleus-firmware/tree/asm
I chose to use X instead of Z in order to leave Z available for future optimizations.  This was a tiny bit more work than using Z since there was a ldd instruction in asmcommon.inc:
https://github.com/micronucleus/micronucleus/blob/master/firmware/usbdrv/asmcommon.inc#L70

cpldcpu, 2021-03-09:
hm... yeah, touching V-USB always opened a whole can of worms. You change one thing and something else pops up.

How do we best approach this? Can we merge V-USB from your repo or are there other incompatibilities?

nerdralph, 2021-03-09:
> 
> 
> hm... yeah, touching V-USB always opened a whole can of worms. You change one thing and something else pops up.
> 
> How do we best approach this? Can we merge V-USB from your repo or are there other incompatibilities?

There are no incompatibilities I am aware of.  I think a few other people should pull my asm branch to do some testing before merging.  The enumeration, erase command, and flashing of a small test sketch with t85_default and _aggressive has been very reliable for me.  I've also done some testing with the t88 (t88_asp config in my fork).
Just yesterday I tried flashing upgrade.hex, and was getting frequent write errors (write error -1).  Obviously this needs further debugging.

nerdralph, 2021-03-09:
Good news:  The flash write errors were not introduced in my asm branch.
Bad news: I've reproduced the same errors with MN 2.5.  Writing a ~1kB image results in flash write errors about 50% of the time.
I've never seen an error with --erase only after dozens of tests.
My current theory is the bug is related to the repeated ~4ms flash write cycles where the MCU is halted.

ArminJo, 2021-03-11:
To clarify: I reactivated the push for Y on 30.1.2021.

nerdralph, 2021-03-12:
> 
> How do we best approach this? Can we merge V-USB from your repo or are there other incompatibilities?

I just finished several hours of testing, including measuring the timing with my scope.  Unfortunately it seems one of my V-USB mods has reduced the reliability vs MN25.  Specifically, using t85_default, my mods reduce the high-side timing error margin.  In other words, when the chosen OSCCAL value is slightly fast, my fork tends to fail where MN25 works OK.  I did additional testing with hard-coded OSCCAL values to be certain.

One good thing I observed in my testing was that tuneOsccal was very reliable with MN25.  The majority of the time it picked an OSCCAL value that was about 0.4% fast (0x5D).  Occasionally it chose the next lower OSCCAL, which was about 0.7% slow (0x5C).

With t85_aggressive it wasn't as good.  The only somewhat reliable OSCCAL value was 0x59, but tuneOsccal would usually pick 0x58 and even sometimes 0x57.  There's a discontinuity between steps when the 3 least significant bits go from 111 to 000, so it's not too surprising.

For testing I used a DIP t85, powered at 3.35V with an AMS1117 regulator on a USBasp.

cpldcpu, 2021-07-15:
I think this was fixed, wasn't it?

closed 2021-07-15 by cpldcpu