micronucleus/micronucleus #228: [BUG] Device detected on Windows 10 only once

alexiri, 2021-03-18:
**Describe the bug**
I'm trying to program a simple mouse emulator with a Chinese Digispark clone. I can program the t85 just fine and it works well on my Linux machine. However, on a Windows 10 laptop it only works the first time it's plugged in.

When I plug it in, I hear the Windows "new device" sound twice (once for the bootloader and once for my mouse code), and I can see that my code is working. However, if I unplug it and plug it in again, Windows completely ignores it. The t85 is working and the LED blinks when it should, but the machine does not see the device at all.

Rebooting does not help and neither does reuploading my code. I can get Windows to see the device again by reflashing the bootloader, but it doesn't always work. Other times I've had to resort to rewriting the flash memory (with avrdude) from a copy I dumped off of a different unmodified unit. I have tried versions 2.2 (factory version), 2.3, 2.5 and master (with AUTO_EXIT_MS set to 10000).

**To Reproduce**
Steps to reproduce the behavior:
1. Upload code
2. Plug in to Windows 10 laptop
3. Windows detects the device and everything works as expected
4. Unplug device
5. Plug device again. Windows now does not detect it.

**Expected behavior**
I expect the device to work correctly every time it's plugged in.

**Used Hardware**
Chinese Digispark clone

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Development environment: digispark programmed 
 - Version of the bootloader [e.g. 2.3]: Same issue with 2.2, 2.3, 2.5 and master

cpldcpu, 2021-03-23:
This seems to be a reproducible enumeration bug, but it's not really clear how to proceed from here for me. 

Since you have already tried several versions of the firmware, this does not seem to be caused by something that was newly introduced. It appears likely that this is specific to your laptop.

Maybe you could try the v1.11 firmware with unmodified V-USB? The old version perform enumartion in a slightly different way, maybe this gives some insights.

Have you tried different ports? 

Btw, what kind of laptop are you using?

alexiri, 2021-03-23:
It's a Lenovo 3rd gen ThinkPad X1 Yoga, and I did try on both of it's USB ports with the same results. It does seem to work if I connect via a USB 2.0 hub, though.

For the v1.11, do you mean [this one](https://github.com/micronucleus/micronucleus/blob/v1.11/firmware/releases/micronucleus-1.11.hex)? If so, I did try but I ended up screwing the device. I must be flashing it incorrectly:

```
$ ./micronucleus ../../../firmware/releases/micronucleus-1.11.hex
> Please plug in the device ... 
> Device is found!
connecting: 40% complete
> Device has firmware version 2.2
> Device signature: 0x1e930b 
> Available space for user applications: 6522 bytes
> Suggested sleep time between sending pages: 7ms
> Whole page count: 102  page size: 64
> Erase function sleep duration: 714ms
parsing: 40% complete
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0xa
> Warning: Checksum error between address 0xa and 0xa
> Warning: Checksum error between address 0x0 and 0xde
> Warning: Checksum error between address 0x0 and 0xde
> Warning: Checksum error between address 0x2302 and 0x235b
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x1
> Warning: Checksum error between address 0x0 and 0x5
> Warning: Checksum error between address 0x0 and 0x1
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0xa and 0x19
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x0 and 0x0
[... 52 lines cut ...]
> Warning: Checksum error between address 0x0 and 0xc
> Warning: Checksum error between address 0xf and 0xf
> Warning: Checksum error between address 0xf and 0xf
> Warning: Checksum error between address 0xf and 0xf
> Warning: Checksum error between address 0x30 and 0x33
> Warning: Checksum error between address 0x0 and 0xc
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0x2302 and 0x235b
> Warning: Checksum error between address 0x0 and 0x24
> Warning: Checksum error between address 0x5 and 0x1d
> Warning: Checksum error between address 0xd and 0xd
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0xd and 0xd
> Warning: Checksum error between address 0xc and 0xc
> Warning: Checksum error between address 0xa and 0xa
> Warning: Checksum error between address 0xc and 0x19
> Warning: Checksum error between address 0x0 and 0xc
> Warning: Checksum error between address 0xd and 0xd
> Warning: Checksum error between address 0x0 and 0xa
> Warning: Checksum error between address 0x0 and 0x0
> Warning: Checksum error between address 0xa and 0xa
> Warning: Checksum error between address 0x0 and 0xab
parsing: 60% complete
> Program file is 2529 bytes too big for the bootloader!
```

cpldcpu, 2021-03-25:
>Program file is 2529 bytes too big for the bootloader!

It seems that your program is too large for the v1.11 bootloader? The older versions are still considerably larger.

serg7461, 2021-06-13:
for using micronucleus command line utility you should use "firmware/upgrades/micronucleus-1.11.hex"
for "/firmware/releases/micronucleus-1.11.hex" use avrdude

closed 2021-07-15 by cpldcpu