micronucleus/micronucleus #239: [REQUEST] support or guide to atmega8

ca4ti, 2021-07-06:
it is possible to install it on ATMEGA8-16PI, If so, how should I proceed or wait for support?

cpldcpu, 2021-07-15:
It should be possible to create a configuration based on the ATMega328 support. As far as I know, nobody is looking into this right now

ca4ti, 2021-07-15:
This discussion ( https://forums.obdev.at/viewtopic.php?t=8291 and this file https://www-user.tu-chemnitz.de/~heha/viewzip.cgi/basteln/PC/USB2LPT/usb2lpt.zip/src/firmware/bootloadHID/src-ATmega8-bootload/Makefile?auto=MAK&lang=en  ) demonstrates that it works with 12.8mhz for v-usb and has the FUSES data and other information.

If it is not bothersome , could you guide me where to put these parameters to test with this atmega8 ?

thanks

cpldcpu, 2021-07-27:
All the required changes are in the configuration file:

https://github.com/micronucleus/micronucleus/blob/master/firmware/configuration/m328p_extclock/Makefile.inc

The ATMega328 is compatible to the ATMega88. Not sure if there are any issues with the ATMega8.

closed 2021-11-04 by ArminJo