micronucleus/micronucleus #244: Digispark won't upload code

FreshImmuc, 2021-09-11:
Hi, I am new to this. This is not a duplicate of #131 but the same issue.  I can't fix it the way he did bc I am on windows and not on linux. I have this config: 
[image]

I get this issue when I try to upload my script:

[image]

I think the digispark gets detected but it's not uploading. The port is only available when I put the board in the port. I also installed all drivers.

ArminJo, 2021-09-12:
You must plug in the board **after** you see the message `Please plug in the device ...`

FreshImmuc, 2021-09-12:
Already tried that..

ArminJo, 2021-11-04:
Have you installed the new [executable ](https://github.com/micronucleus/micronucleus/tree/master/commandline/builds/x86_64-mingw32)?
And the new [board definitions](https://github.com/ArminJo/micronucleus-firmware#updated-configuration-for-digispark-boards)?