micronucleus/micronucleus #247: cant get usb working attiny167

hitchhiker85, 2021-09-20:
hi.
i make it short...

i bought on aliexpress some attiny167pro boards.

when i connect to usb i got failure : that the vendor is not reading. yellow triangle with exclamation mark.

then i install libusb... nothing happend.
i reburn via usbasp the default 167 hex.
nothing happend.

i can programm with usbasp but with the icrousb port i can do nothing.

then i wired to pb3 and pb6 d lines from a usb caple and gnd and 5v.. nothing happend.

what is the failure?!

thanks

cpldcpu, 2021-09-20:
That sounds like something that the vendor of these boards should support in. Usually the configuration of the bootloader has to be adopted for a specific hardware.

hitchhiker85, 2021-09-20:
mmmh ok.. any idea or tip to fix it?!

ArminJo, 2021-11-04:
Try another **Cable**, try another **Port**, try another **Board**, try another **Computer**, try another **OS**.
In this order.