micronucleus/micronucleus #267: Sometime after upgrade to 2.6 device uresponsive

Barabba11, 2022-07-27:
Hi everyone, maybe it' s a electrical fault of CPU, but it's hard for me to believe that, since it was already boxed and connected to the same DC supply @5,08V. There may be a spike during powerup.. it's hard for me to believe that is enough to burn the controller, so I more suppose it's a problem of Bootloader.
It was version 1.6 (that showed micronucleus /? ), it worked ok for about 10 times (power on) and not for long time. The last working time it waited the common 6 seconds before the sketch loaded, all ok, now doesn't react to the USB recognition, not running the sketch. 
I've tried to connect PB6 (reset pin) directly to VCC, I removed all connection (free board) and I tried to power it up with all pins to VCC, no progress.
How can I know if it's completely dead?

Thanks

ArminJo, 2022-07-28:
How did you know that it is a bug of the micronucleus?
Ever heard if ISP?

closed 2022-07-28 by ArminJo

Barabba11, 2022-07-28:
I'm just reporting it in case there are evidence of other faults, you can't exclude it's a problem of bootloader as well. 
The operationg conditions if ATtiny85 were ideal, and for me it's hard to believe the controller gets broken.

I'm asking a way to test it and know more, how?
Thanks for who will help me go deeper

cpldcpu, 2022-07-28:
HI Barraba11,

thanks for reporting, but based on your description it is really hard to discern what is going on. As Armin pointed out, the best thing would be to use an ISP to check if the device is still alive and whether the bootloader was corrupted.

Barabba11, 2022-07-28:
Thanks for answer, sincerely I've never searched/viewed an ISP for attiny85, I mean if it is invasive for Digispark and it means I need to solder out the IC from the board (I suppose that due to zerner diodes). I'm asking if there is a no invasive way, I suppose there isn't any. So the only think I can do is to ship this Digispark to you, and hope you may understand how it briked, and maybe improve BL if that needs..

ArminJo, 2022-07-28:
@Barabba11 Please first read, then judge.

cpldcpu, 2022-07-28:
@Barabba11, regarding Digispark support, it would be best to use their forum. Micronucleus is an independent project supported by volunteers in their free time.

Barabba11, 2022-07-28:
Well, Digispark in my opinion means only a couple of zerner with resistors placed on 2 pins of MCU,. I can write there too..
Chances are, if the fault it's up to BL, I'm in the right place to understand and let you know my experience. I don't pretend nothing from  anyone, I can trash it and look forward, if you think it's better do that