micronucleus/micronucleus #274: Digispark loses firmware after power  off and on

Portia-Lin, 2023-02-18:
Digispark loses firmware
I have a problem with digispark board. Brief description:
My sketch controls the servo via button presses. Powering the circuit from the battery through the step-up module MT3608. The TP4056 battery charging and protection board is also used. The battery itself may also have a built-in protection board, it could not be verified. The sketch works correctly.
The problem is that when the servo hits an obstacle, it takes a lot of effort to move further and loses power. Here, one or more protections are activated and the power supply is probably lost for a while. In theory, after power is restored, the controller should start the program execution from the beginning. But he doesn't do that. Only flashing the microcontroller helps.
Why does this happen and how to overcome it?
Microcontroller - Digispark Attiny85 with micro USB input.

cpldcpu, 2023-02-18:
That sounds like a hardware issue with the microcontroller, related to supply glitching, not really related to micronucleus? I would suggest to ask in an AVR forum.

closed 2023-02-18 by cpldcpu

Portia-Lin, 2023-06-04:
Not hardware problem. This is software probleb with Chinese digispark. In fuses bodlevel doesn't set and bootloader rewrite the program because voltage decrease