micronucleus/micronucleus #276: [BUG] Setting FAST_EXIT_NO_USB_MS breaks compilation

lestrozi, 2023-09-16:
Please also have a look at discussions: https://github.com/micronucleus/micronucleus/discussions

**Describe the bug**
Setting FAST_EXIT_NO_USB_MS to any value different than 0 (even <120) breaks compilation
I assume this is related to the fact that setting this config adds 8 bytes, probably shifting something unexpectedly

**To Reproduce**
Steps to reproduce the behavior:
1. Change configuration/t85_default/bootloaderconfig.h:237 to "#define FAST_EXIT_NO_USB_MS       120"
2. make
4. See error:
```
$ make
/usr/bin/avr-ld: address 0x200a of main.bin section `.text' is not within region `text'
/usr/bin/avr-ld: address 0x200a of main.bin section `.text' is not within region `text'
collect2: error: ld returned 1 exit status
make: *** [Makefile:97: main.bin] Error 1
```

**Used Hardware**
t85_default

**Desktop (please complete the following information):**
 - OS: Linux
 - Version of the bootloader: 2.6

lestrozi, 2023-09-16:
Well, nevermind, that was fixed by setting `BOOTLOADER_ADDRESS = 19C0`

closed 2023-09-16 by lestrozi

cpldcpu, 2023-09-16:
Indeed, that is the correct way of handling the issue.