micronucleus/micronucleus #277: micronucleus flashing but not running user app

zinahe, 2023-10-30:
Hi, 

I have successfully built micronucleus for ATtiny44A with an external 12Mhz crystal clock using the following settings and flashed it to the MCU.

Makefile.inc
```
F_CPU = 12000000
DEVICE = attiny44

# 4kb = 1024 * 4 = 4096
# 4096 - 1444 = 2652 = 0xA40 (2652 / 64 = 41)
BOOTLOADER_ADDRESS = A40

FUSEOPT = -U lfuse:w:0xef:m -U hfuse:w:0xdd:m -U efuse:w:0xfe:m

#---------------------------------------------------------------------
# ATtiny44
#---------------------------------------------------------------------
# Fuse extended byte:
# 0xFE = - - - -   - 1 1 0
#                        ^
#                        |
#                        +---- SELFPRGEN (enable self programming flash)
#
# Fuse high byte:
# 0xdd = 1 1 0 1   1 1 0 1
#        ^ ^ ^ ^   ^ \-+-/ 
#        | | | |   |   +------ BODLEVEL 2..0 (brownout trigger level -> 2.7V)
#        | | | |   +---------- EESAVE (preserve EEPROM on Chip Erase -> not preserved)
#        | | | +-------------- WDTON (watchdog timer always on -> disable)
#        | | +---------------- SPIEN (enable serial programming -> enabled)
#        | +------------------ DWEN (debug wire enable)
#        +-------------------- RSTDISBL (disable external reset -> enabled)
#
# Fuse low byte:
# 0xef = 1 1 1 0   1 1 1 1
#        ^ ^ \+/   \--+--/
#        | |  |       +------- CKSEL 3..0 (clock selection -> XTAL Oscillator)
#        | |  +--------------- SUT 1..0 (BOD enabled, fast rising power)
#        | +------------------ CKOUT (clock output on CKOUT pin -> disabled)
#        +-------------------- CKDIV8 (divide clock by 8 -> don't divide)
```
When I try to flash a simple program to toggle a GPIO pin it appears to complete without issues; with micronucleus reporting success.

```
> Device is found!
connecting: 33% complete
> Device has firmware version 2.3
> Device signature: 0x1e9207
> Available space for user applications: 2618 bytes
> Suggested sleep time between sending pages: 7ms
> Whole page count: 41  page size: 64
> Erase function sleep duration: 287ms
parsing: 50% complete
> Erasing the memory ...
erasing: 66% complete
> Starting to upload ...
writing: 83% complete
> Starting the user app ...
running: 100% complete
>> Micronucleus done. Thank you!
```

But nothing happens at the pin when I power-cycle the MCU. At first, I thought there might be a problem with the test program. But when I flash the very same hex file using AVR ISP directly to the Attiny44 it works. It's like micronucleus not not loading/running the user program on reset. 

Any help is greatly appreciated.

Thanks

zinahe, 2023-10-30:
This is what I have in bootloaderconfig.h if helps in anyway:

```
#define USB_CFG_IOPORTNAME      A
#define USB_CFG_DMINUS_BIT      0
#define USB_CFG_DPLUS_BIT       7
#define USB_CFG_CLOCK_KHZ       (F_CPU/1000)

// setup interrupt for Pin Change for D+
#define USB_INTR_CFG            PCMSK0
#define USB_INTR_CFG_SET        (1 << USB_CFG_DPLUS_BIT)
#define USB_INTR_CFG_CLR        0
#define USB_INTR_ENABLE         GIMSK
#define USB_INTR_ENABLE_BIT     PCIE0
#define USB_INTR_PENDING        GIFR
#define USB_INTR_PENDING_BIT    PCIF0
#define USB_INTR_VECTOR         PCINT0_vect
    
#define MICRONUCLEUS_WRITE_SLEEP 5

// ATtiny84 does not know WDTCR
#ifndef WDTCR
#define WDTCR WDTCSR
#endif

#define ENTRYMODE ENTRY_ALWAYS

#define ENTRY_ALWAYS    1
#define ENTRY_WATCHDOG  2
#define ENTRY_EXT_RESET 3
#define ENTRY_JUMPER    4

#if ENTRYMODE==ENTRY_ALWAYS
  #define bootLoaderInit()
  #define bootLoaderExit()
  #define bootLoaderStartCondition() 1
#endif

#define AUTO_EXIT_NO_USB_MS    0
#define AUTO_EXIT_MS           6000
 
 
#define OSCCAL_RESTORE_DEFAULT 0
#define OSCCAL_SAVE_CALIB 0
#define OSCCAL_HAVE_XTAL 1
  
#define LED_MODE    NONE
```

Sojourneer, 2023-11-10:
I'm currently having a problem which I think is related to using the D- connected MCU pin for something else at runtime (a WS2812B data line).  I disconnect it from the USB D- after flashing, but it seems to confuse the bootloader.

cpldcpu, 2023-11-10:
So the line is connected to a WS2812 DIN? That may be too much load for the USB. Not sure it is related to the bootloader.

zinahe, 2023-11-15:
Nevermind, I figured it out after a lot of banging my head on the keyboard. I didn't know I had to flash the fuses separately. The `efuse 0xFE` was very important or else the ATTINY44 will not be able to program it's own flash without it. I was only flashing the firmware binary all along.

Thanks.

zinahe, 2023-11-15:
It's working now.

closed 2023-11-15 by zinahe